socallinuxexpo/scale-network #914: Pi case HDMI cable fit problem

ptrlow1, 2025-03-06:
Some (not all) of our Pi cases may not fit some micro hdmi cables.
The red microhdmi cables fits, but the black microhdmi cables don't fit, and have some trouble getting a good connection and being recognized by the tv.

ptrlow1, 2025-03-06:
Case it on the pi could be aligned (top pi)
Several have Pis that have the board (bottom pi) and some micro HDMI cables may not fit and get a firm connection.

[image]

ptrlow1, 2025-03-06:
M3x6mm screws should be used on securing to bottom out the case.
M3x8mm screws are used when using the VESA mount.
"@owendelong"

tweak42, 2025-09-28:
Moved 6mm screws to bottom plate and 8mm screws for tops.  There is a small gap between the top standoff and top plate.  All Pi4 corrected.

sarcasticadmin, 2025-09-28:
Thank you @tweak42 and @ptrlow1 much appreciated!

closed 2025-09-28 by sarcasticadmin