socallinuxexpo/scale-network #920: Get a bigger Internet circuit

MrHamel, 2025-03-07:
With the new OpenWrt One access points doing so well, we've saturated the 800Mbit AT&T uplink provided by the convention center, resulting in Owen adding policers to various switches.

This issue is to start the conversation on getting a bigger circuit (>= 2Gbit) from the convention center (via AT&T or Encore's new vendor), Cogent, or Crown Castle.

sarcasticadmin, 2025-12-04:
I got an update from PCC about the new circuit: `Provider is Lumen, same 1G speed, and unfortunately no native ipv6`

MrHamel, 2025-12-04:
Not surprising. Not many events use that connectivity, so we can't blame them for not the level of connectivity we need for one week of a year.

sarcasticadmin, 2025-12-10:
Closing since theres not much else we can do