socallinuxexpo/scale-network #929: snmp.yaml generator drv in monitoring

sarcasticadmin, 2025-03-07:
Related to: #https://github.com/socallinuxexpo/scale-network/pull/928

The snmp.yaml was generated using https://github.com/prometheus/snmp_exporter/tree/v0.28.0/generator paired down the just the juniper stats we care about. This should be a drv instead of being vendored in scale-network

 ## AC:

- make the snmp.yaml generator a drv in the monitoring service

cjmakes, 2025-03-07:
Started working on this here:https://github.com/socallinuxexpo/scale-network/blob/conjones/929/nix/nixos-modules/services/prometheus/snmp.nix

But I've hit my limit of nix config and could use a hand figuring out how to get the mibs into the build context. It looks like `fetchurl` is trying to execute the file which is downloaded 
```
prometheus>nix-build -E 'with import <nixpkgs> {}; callPackage ./snmp.nix {}'
this derivation will be built:
  /nix/store/jwqc904dcrg202601vmnx3ysag9y9gdm-snmp-configuration.drv
building '/nix/store/jwqc904dcrg202601vmnx3ysag9y9gdm-snmp-configuration.drv'...
/nix/store/xfn7gny1w7yrqgijf386zbcawqqnpcrb-IANA-IFTYPE-MIB.txt: line 1: IANAifType-MIB: command not found
error: builder for '/nix/store/jwqc904dcrg202601vmnx3ysag9y9gdm-snmp-configuration.drv' failed with exit code 127;
       last 1 log lines:
       > /nix/store/xfn7gny1w7yrqgijf386zbcawqqnpcrb-IANA-IFTYPE-MIB.txt: line 1: IANAifType-MIB: command not found
       For full logs, run 'nix log /nix/store/jwqc904dcrg202601vmnx3ysag9y9gdm-snmp-configuration.drv'.
```

BerkhanBerkdemir, 2025-03-09:
`snmp.yml` generation is only required if we are changing our interest in OIDs or in this case "modules".  In order to collect some of the _interesting_ OIDs, we still needed to change the upstream `Makefile` to find all the standard and vendor specific OIDs.