socallinuxexpo/scale-network #934: LLDP not reporting from signs

owendelong, 2025-03-07:
The signs are not reporting LLDP

owendelong, 2025-03-07:
FYI:
EmitLLDP=
Controls support for Ethernet LLDP packet emission. Accepts a boolean parameter or the special values "nearest-bridge", "non-tpmr-bridge" and "customer-bridge". Defaults to false, which turns off LLDP packet emission. If not false, a short LLDP packet with information about the local system is sent out in regular intervals on the link. The LLDP packet will contain information about the local hostname, the local machine ID (as stored in [machine-id(5)](https://www.freedesktop.org/software/systemd/man/latest/machine-id.html#)) and the local interface name, as well as the pretty hostname of the system (as set in [machine-info(5)](https://www.freedesktop.org/software/systemd/man/latest/machine-info.html#)). LLDP emission is only available on Ethernet links. Note that this setting passes data suitable for identification of host to the network and should thus not be enabled on untrusted networks, where such identification data should not be made available. Use this option to permit other systems to identify on which interfaces they are connected to this system. The three special values control propagation of the LLDP packets. The "nearest-bridge" setting permits propagation only to the nearest connected bridge, "non-tpmr-bridge" permits propagation across Two-Port MAC Relays, but not any other bridges, and "customer-bridge" permits propagation until a customer bridge is reached. For details about these concepts, see [IEEE 802.1AB-2016](https://standards.ieee.org/findstds/standard/802.1AB-2016.html). Note that configuring this setting to true is equivalent to "nearest-bridge", the recommended and most restricted level of propagation. See LLDP= above for an option to enable LLDP reception.

Added in version 230.

Based on this, any of true, non-tpmr-bridge, or nearest-bridge would work I recommend non-tpmr-bridge for our environment.

sarcasticadmin, 2026-01-29:
Example of enabling this: https://github.com/socallinuxexpo/scale-network/blob/8e80ec347a019408fe42387566192f99c6df64f8/nix/nixos-configurations/core-master/default.nix#L77-L81